FrankBoesing/Arduino-Teensy-Codec-lib #5: audio.h location

wdonno, 2016-01-23:
Hi Frank..
I was trying to compile your tetris for Teensy and ran into problems with the flac codec, where it calls for "audio.h", but it is not being found.

Arduino-Teensy-Codec-lib/FLAC/stream_decoder.c:59:#include "audio.h"

Where should I be sourcing that?

I am working on a linux system and had some other minor problems compiling, due to inconsistent capitalization of the directory name FLAC/flac.  I manually made them consistent.  As a beginner, if there was a better way to fix those file not found errors, I would appreciate any pointers!
[Tetris is a bit of a diversion, mostly trying to learn more about DSP and the teensy audio library]. 

Thanks for any help.
Best wishes,
Bill.

FrankBoesing, 2016-01-23:
Hi, i think it is Audio.h from the Audio Library. Can't verify this at the moment, because im on the road.
Thank you for reporting the capitalization issue. Can you post a list, or better, do a pullrequest? 
Thank you, Frank

FrankBoesing, 2016-01-23:
I've updated two files and edit audio.h to Audio.h 
Are there more issues like that ?

FrankBoesing, 2016-01-24:
I'm closing this, if you find more issures, pls open a new issue.
Thanks!

closed 2016-01-24 by FrankBoesing

wdonno, 2016-01-25:
Thanks.. so with the Audio.h capitalization I was able to compile without error.
The main issue was mixed capitalization for references such as:#include "flac/ordinals.h" or
#include "FLAC/assert.h"

 Arbitrarily I solved by standardizing on FLAC, which meant changing the capitalization of the directory structure.  Now  realize that changing the directory name from flac to FLAC is considered by git to be deleting the former directory and adding a new directory... this may be too big a change for you.
I am happy to go back and re-do it by making all references to "flac", and could do this later in the week.
Let me know if you want me to "make a pull request"... is that correct term? with what i have now or wait until I have reversed the capitalization back to lower case.
Best wishes,Bill.